EECS581Team15/WifiLMR #2: UI design

zjcers, 2018-10-08:
* Create mock ups for the various user interface screens.
   Make some pictures in Photoshop or GIMP as reference for when we're implementing the UI.
* Document user work flows
   Diagrams for how a user uses the UI (ex: for this task, press <button> to reach this screen, or select the given menu item)

jackdweber, 2018-10-17:
UI Designs needed
* Main Screen
* Settings / Applications screen
    * Possible theming 
    * WiFi Selection?
    * Volumes
    * More to come
* Channel selection
* Messaging Screen
    * Messaging Detail Screen

zjcers, 2018-10-18:
Volume and channel selection will have hardware controls. We'll probably want a selection screen for channels either way, though. Might want a nice display slider to pop up when adjusting volume, too.

I think "zones" (groups of channels) were in the spec, so we'll probably want a list screen for that, too.

zjcers, 2018-10-22:
Note: the screen we selected has a resolution of 160x128.

AstralA, 2018-10-29:
[image] Main Screen
[image] Setting Screen
[image] Channel Screen
[image] Messages Screen
[image] Zones Screen
[image] User Workflow

closed 2018-10-29 by AstralA